YosysHQ/picorv32 #261: how to run dhrystone benchmark

toolchainman, 2024-05-17:
Hi, I'm writing to ask for advice. 
I came across github while looking for a benchmark to utilize. Since I am new to everything, I had a lot of difficulties starting with the installation of ubuntu. Eventually, I completed the installation and found dhrystone in the picorv32 folder. I even checked the dhrystone benchmark result of the picorv processor through the make command, but I have no idea how to check the performance of my processor. My processor is made of rtl code. I am looking for help on this.

nerdylye, 2025-03-03:
Hey @toolchainman , I'm currently facing the same issue you had. I'm trying to run the Dhrystone benchmark on another riscv processor (VexRiscv). May I know if you have the solutions to it already? Any help would be much appreciated

toolchainman, 2025-03-04:
I'm currently having some issues with the hardware I designed, so I've put benchmark testing on hold for now. However, if you refer to the readme file and install it, you should have dhrystone benchmark installed. As I recall, it is split into several files like main.h dhrystone.c and you can convert it using gcc compiler and put it on another processor. 
Hope this helps.

nerdylye, 2025-03-05:
Hi thanks for the reply. Yeah I actually compiled the dhrystone files and turn them into a hex file and load it into VexRiscv processor. It was able to run but I'm not getting the Dhrystone benchmark outputs from my terminal/console so I'm not sure if the benchmark is running correctly or not. FYI I'm testing it without any hardware.

nerdylye, 2025-03-06:
Hmm, what I meant was I couldn't get the Dhrystone startup message in the terminal like this even though it ran without errors. 

[image]

For your compilation, according to the sample in this repo `/dhrystone/Makefile`, you just need to modify the `TOOLCHAIN_PREFIX `and `CFLAGS` to `rv32i` and run `make dhry.hex` if you want to compile into a hex file.

toolchainman, 2025-03-06:
I haven't done this yet, just like you, so I can't help with this problem. But I also need to solve it, so if I manage to do so, I'll leave a comment. It would be great if we could help each other

thanks